sci-bots/microdrop #12: Linux and mac release (Trac #8)

cfobel, 2016-10-26:
Migrated from http://microfluidics.utoronto.ca/ticket/8
```json
{
    "status": "new", 
    "changetime": "2014-03-13T20:00:48", 
    "description": "All of the project code has been written as much as possible using cross-platform code and modules, although so far I have only tried running things under a Windows environment. In future, it would be nice to expand to other platforms including Mac and Linux.\n\nOne potential stumbling block is the Serial code in the [http://microfluidics.utoronto.ca/dmf_control_board dmf_control_board library].  It is based on the [http://www.boost.org/doc/libs/1_47_0/doc/html/boost_asio.html boost asio library] which [https://gitorious.org/serial-port/serial-port/blobs/master/3_async/AsyncSerial.cpp may have some problems] on the Mac platform.\n\nIf we do in fact move to supporting releases beyond just Windows, it would probably make sense to migrate from [http://www.py2exe.org/ py2exe] to a cross-platform tool like [http://www.pyinstaller.org/ PyInstaller].\n\n!PyInstaller has the following features:\n * Multiplatform: works under Windows (32-bit and 64-bit), Linux (32-bit and 64-bit) and Mac OS X (32-bit only, 64-bit in svn, see [http://www.pyinstaller.org/wiki/MacOsCompatibility MacOsCompatibility]).\n * Multiversion: works under any version of Python from 2.2 up to 2.7. \n * possible to script [http://groups.google.com/group/pyinstaller/browse_thread/thread/bdeab706503f067b?pli=1 \"generation of Windows executables in Linux using wine\"]", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394740848122658", 
    "component": "core", 
    "summary": "Linux and mac release", 
    "priority": "minor", 
    "keywords": "linux, mac, cross-platform", 
    "version": "", 
    "time": "2011-10-05T04:56:59", 
    "milestone": "", 
    "owner": "ryan", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20111009T16:10:16: cfobel commented:

> Confirmed that microdrop with dmf_control_board_library runs in Ubuntu 11.04 x86, with Boost and GTK libraries installed.

cfobel, 2016-10-27:
Trac update at 20111009T16:11:12: cfobel commented:

> Ubunut 11.04 x86:  Arduino Mega is recognized properly.

cfobel, 2016-10-27:
Trac update at 20121018T18:35:30: ryan changed milestone from "" to "0.3"

cfobel, 2016-10-27:
Trac update at 20140313T20:00:48: ryan changed milestone from "1.1" to ""

ryanfobel, 2017-01-05:
This issue will be covered by issues in the [Linux release](https://github.com/wheeler-microfluidics/microdrop/labels/cat%3ALinux%20release) category.

closed 2017-01-05 by ryanfobel