sci-bots/microdrop #113: Add video recording using GStreamer (Trac #109)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/microdrop/ticket/109
```json
{
    "status": "assigned", 
    "changetime": "2014-03-13T20:00:48", 
    "description": "", 
    "reporter": "cfobel", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394740848122658", 
    "component": "", 
    "summary": "Add video recording using GStreamer", 
    "priority": "major", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-06-23T19:56:32", 
    "milestone": "", 
    "owner": "cfobel", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20120801T15:58:43: ryan changed status from "new" to "assigned"

cfobel, 2016-10-27:
Trac update at 20140313T20:00:48: ryan changed milestone from "1.1" to ""

ryanfobel, 2017-01-05:
The GTK device view is going to be deprecated in favor of the [webui-plugin](https://github.com/wheeler-microfluidics/webui-plugin).

closed 2017-01-05 by ryanfobel