sci-bots/microdrop #154: Document creating update repostories (Trac #150)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/microdrop/ticket/150
```json
{
    "status": "new", 
    "changetime": "2014-03-13T19:12:22", 
    "description": "Write documentation for creating a new update repository, including apache config, creating a django database, etc. Also document how to add additional repository instances to an existing apache config.", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394737942842884", 
    "component": "", 
    "summary": "Document creating update repostories", 
    "priority": "minor", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2014-03-13T19:12:22", 
    "milestone": "", 
    "owner": "", 
    "type": "task"
}
```

ryanfobel, 2017-01-05:
This ticket description should be updated once we decide on a new plugin update repository. We have discussed converting MicroDrop into conda packages and using conda channels as update repositories (with a `conda-helpers` package that provides an API to query and install plugins).

ryanfobel, 2017-01-05:
This ticket description should be updated once we decide on a new plugin update repository. We have discussed converting MicroDrop into conda packages and using conda channels as update repositories (with a `conda-helpers` package that provides an API to query and install plugins). Until this issue is blocked.