sci-bots/microdrop #53: Add ability to plot current, force and actuation voltage (Trac #49)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/ticket/49
```json
{
    "status": "new", 
    "changetime": "2014-04-17T19:40:13", 
    "description": "\n", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1397763613851119", 
    "component": "microdrop", 
    "summary": "Add ability to plot current, force and actuation voltage", 
    "priority": "minor", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-02-20T03:47:13", 
    "milestone": "Microdrop 1.1", 
    "owner": "ryan", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20120220T04:04:55: ryan changed summary from "Add ability to plog current, force and actuation voltage" to "Add ability to plot current, force and actuation voltage"

cfobel, 2016-10-27:
Trac update at 20140417T19:40:13: ryan changed component from "gui" to "microdrop"

ryanfobel, 2017-01-05:
This was implemented a long time ago in the [dmf-control-board-plugin](https://github.com/wheeler-microfluidics/dmf_control_board_plugin).

closed 2017-01-05 by ryanfobel