sci-bots/microdrop #56: Add option to display feedback threshold on graphs (Trac #52)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/ticket/52
```json
{
    "status": "new", 
    "changetime": "2014-03-13T20:00:48", 
    "description": "", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394740848122658", 
    "component": "core", 
    "summary": "Add option to display feedback threshold on graphs", 
    "priority": "minor", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-02-20T03:55:37", 
    "milestone": "", 
    "owner": "ryan", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20140313T20:00:48: ryan changed milestone from "1.1" to ""

ryanfobel, 2017-01-05:
Migrated this issue to the [DMF control board plugin issue tracker](https://github.com/wheeler-microfluidics/dmf_control_board_plugin/issues/3).

closed 2017-01-05 by ryanfobel