sci-bots/microdrop #57: Add tool for cycling through electrodes on device (Trac #53)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/ticket/53
```json
{
    "status": "accepted", 
    "changetime": "2014-04-17T19:40:13", 
    "description": "This could be used to check that all electrodes are connected (i.e., measure capacitance of air gap) and/or to figure out on startup which reservoirs contain liquid.", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1397763613851119", 
    "component": "microdrop", 
    "summary": "Add tool for cycling through electrodes on device", 
    "priority": "minor", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-02-20T03:56:43", 
    "milestone": "Microdrop 1.1", 
    "owner": "ryan", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20120223T04:56:38: ryan changed description from "This could be used to check that are all electrodes are connected (i.e., measure capacitance of air gap) and/or to figure out on startup which reservoirs contain liquid." to "This could be used to check that all electrodes are connected (i.e., measure capacitance of air gap) and/or to figure out on startup which reservoirs contain liquid."

cfobel, 2016-10-27:
Trac update at 20120302T02:02:49: ryan changed status from "new" to "accepted"

cfobel, 2016-10-27:
Trac update at 20140417T19:40:13: ryan changed component from "core" to "microdrop"

ryanfobel, 2017-01-05:
Migrated to https://github.com/wheeler-microfluidics/dmf_control_board_plugin/issues/5

closed 2017-01-05 by ryanfobel