sci-bots/microdrop #58: Calibration wizard to measure feedback resistance/capacitance (Trac #54)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/ticket/54
```json
{
    "status": "accepted", 
    "changetime": "2014-04-17T19:40:13", 
    "description": "The control board has a bank of resistors used to measure device impedance and amplifier voltage. These resistors need to be characterized in order to get accurate measurements.\n\nIn addition, each resistor has an associated parasitic capacitance due to the cables, PCB traces, etc., which also need to be accounted for.\n\nWe should define a test load(s) that can be attached to the control board and have a wizard that will walk a user through a series of steps, collect the necessary data, fit curves and assign values to the calibration variables.", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1397763613851119", 
    "component": "microdrop", 
    "summary": "Calibration wizard to measure feedback resistance/capacitance", 
    "priority": "major", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-02-20T04:03:01", 
    "milestone": "Microdrop 1.1", 
    "owner": "ryan", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20120310T16:05:25: ryan changed status from "new" to "accepted"

cfobel, 2016-10-27:
Trac update at 20140417T19:40:13: ryan changed component from "core" to "microdrop"

ryanfobel, 2017-01-05:
This was implemented a long time ago...

closed 2017-01-05 by ryanfobel