sci-bots/microdrop #61: Video frame updates are not processed while plugins are blocking (Trac #57)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/ticket/57
```json
{
    "status": "assigned", 
    "changetime": "2014-03-13T20:00:48", 
    "description": "When plugins are handling emit_signal requests they are blocking the main thread; therefore video frames are not updated. We need to figure out a way to update video frames (and maybe be responsive to gui events?) while plugins are handling signals.\n\nOne possible solution may be to set up two different types of emit_signal calls (blocking and non-blocking). Non blocking calls would spawn a new thread (or communicate with a concurrently running thread/process). These non-blocking signal handlers would need to be made thread-safe.", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394740848122658", 
    "component": "gui", 
    "summary": "Video frame updates are not processed while plugins are blocking", 
    "priority": "major", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-02-20T04:26:04", 
    "milestone": "", 
    "owner": "cfobel", 
    "type": "enhancement"
}
```

cfobel, 2016-10-27:
Trac update at 20120309T14:42:14: 
* ryan changed description from:

> The video capture appears to lead the electrode actuation by a single frame. While this is not really a major problem, it may be a bit confusing to the user.

to:

> When plugins are handling emit_signal requests they are blocking the main thread; therefore video frames are not updated. We need to figure out a way to update video frames (and maybe be responsive to gui events?) while plugins are handling signals.
> 
> One possible solution may be to set up two different types of emit_signal calls (blocking and non-blocking). Non blocking calls would spawn a new thread (or communicate with a concurrently running thread/process). These non-blocking signal handlers would need to be made thread-safe.

* ryan changed milestone from "0.2" to "0.3"
* ryan changed priority from "trivial" to "major"
* ryan changed summary from "Video appears to be 1 frame ahead of actuation" to "Video frame updates are not processed while plugins are blocking"
* ryan changed type from "defect" to "enhancement"

cfobel, 2016-10-27:
Trac update at 20120714T15:55:12: ryan@fobel.net commented:

> commit 8edbf05717049c7756b51aa366849f3ab7ca5922
> Author: Ryan Fobel <ryan@fobel.net>
> Date:   Sat Jul 14 11:52:38 2012 -0400
> 
>     Process gtk events between emitting signals
>     
>     This shoud make the gui more responsive. Note that threads that will be blocking
>     for significant amounts of time should also be calling gtk.main_iteration()
>     intermitently.
>     
>     See #61.

cfobel, 2016-10-27:
Trac update at 20120714T15:56:54: ryan commented:

> I'm not actually convinced that this problem has anything to do with video, as there is a similar lag in updating electrode state even when the video is disabled. Instead, I think there might be a problem with gtk updating of the dmf_device_view widget.
> 
> Other gtk widgets (e.g., the protocol_list_view) seem to update as expected while a protocol is running, which means this is independent of the blocking threads issue.

cfobel, 2016-10-27:
Trac update at 20120801T15:56:40: 
* ryan changed milestone from "0.3" to "0.2"
* ryan changed owner from "ryan" to "cfobel"
* ryan changed status from "new" to "assigned"

cfobel, 2016-10-27:
Trac update at 20140313T20:00:48: ryan changed milestone from "1.1" to ""

ryanfobel, 2017-01-05:
Video is now going to be handled by the [dmf_device_ui_plugin](https://github.com/wheeler-microfluidics/dmf_device_ui_plugin) which is outside of the MicroDrop core and since it is not running in the GTK loop, this issue no longer exists.

closed 2017-01-05 by ryanfobel