sci-bots/microdrop #85: Change label to indicate that 'duration' is actually minimum duration (Trac #81)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/microdrop/ticket/81
```json
{
    "status": "reopened", 
    "changetime": "2014-03-13T20:00:48", 
    "description": "", 
    "reporter": "cfobel", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394740848122658", 
    "component": "gui", 
    "summary": "Change label to indicate that 'duration' is actually minimum duration", 
    "priority": "major", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-03-24T00:06:53", 
    "milestone": "", 
    "owner": "ryan", 
    "type": "task"
}
```

cfobel, 2016-10-27:
Trac update at 20120329T17:00:47: 
* ryan@fobel.net commented:

> commit 51e8b0e4764f714f3fd511dd5e56faddff43aa80
> Author: Ryan Fobel <ryan@fobel.net>
> Date:   Thu Mar 29 10:49:05 2012 -0400
> 
>     Update GUI re: #84, #85
>     
>     Fixes #84 by emphasizing that the Voltage, Frequency, Duration parameters relate
>     to the current step. Rename "duration" to "minimum duration" (#85) but to close #85,
>     we should also rename the duration member of DmfControlBoardOptions so that it
>     is also reflected in the protocol grid, or better yet, pull the duration
>     parameter out of the dmf_control_board plugin since it is actually more general.

* ryan@fobel.net changed resolution from "" to "fixed"
* ryan@fobel.net changed status from "new" to "closed"

cfobel, 2016-10-27:
Trac update at 20120330T01:25:48: 
* ryan changed resolution from "fixed" to ""
* ryan changed status from "closed" to "reopened"

cfobel, 2016-10-27:
Trac update at 20140313T20:00:48: ryan changed milestone from "1.1" to ""

ryanfobel, 2017-01-05:
This issue will probably become irrelevant once we have a new protocol format #204.

closed 2017-01-05 by ryanfobel