sci-bots/microdrop #92: Add Ubutnu buildbot(s) (Trac #88)

cfobel, 2016-10-27:
Migrated from http://microfluidics.utoronto.ca/microdrop/ticket/88
```json
{
    "status": "new", 
    "changetime": "2014-03-13T20:00:48", 
    "description": "Either one machine supporting the latest version or multiple VMs supporting the most recent plus LTS.", 
    "reporter": "ryan", 
    "cc": "", 
    "resolution": "", 
    "_ts": "1394740848122658", 
    "component": "core", 
    "summary": "Add Ubutnu buildbot(s)", 
    "priority": "major", 
    "keywords": "", 
    "version": "0.1", 
    "time": "2012-03-30T20:16:26", 
    "milestone": "", 
    "owner": "ryan", 
    "type": "defect"
}
```

cfobel, 2016-10-27:
Trac update at 20140313T20:00:48: ryan changed milestone from "1.1" to ""

ryanfobel, 2017-01-05:
Closing since we no long support a Linux build. Once we do, we will probably switch to [Travis](https://travis-ci.org/) anyway.

closed 2017-01-05 by ryanfobel