sci-bots/microdrop #188: `dmf_control_board_plugin` plugin errors with DropBot v3.0

cfobel, 2016-11-07:
The `dmf_control_board_plugin` plugin is only valid for the old DropBot v2.0 hardware and will prevent the DropBot v3.0 from operating correctly.

# Workaround

If you can start MicroDrop, you can uninstall the `dmf_control_board_plugin` plugin through the `File>Plugin manager` menu.  If MicroDrop is crashing before you have a chance you can manually delete the `dmf_control_board_plugin` plugin using the following steps:

 - Close MicroDrop if it is running.
 - Browse to `Documents\MicroDrop\plugins`
 - Delete the `dmf_control_board_plugin` directory

cfobel, 2016-11-07:
We should probably try to figure out a more general way to solve this issue.  It will especially be a problem for computers used to control both DropBot v2.0 and DropBot v3.0 (not at the same time).

ryanfobel, 2016-11-07:
Instead of deleting the plugin, you can also just disable it; but I agree that we should figure out a better way to handle support for multiple types of hardware.