sci-bots/microdrop #190: ImportError: No module named _backend_gdk

ryanfobel, 2016-11-09:
The error is triggered when importing the `dmf_control_board_plugin` in MicroDrop v2.1. Here's the full traceback:

```python
2016-11-09 18:16:53 [INFO:root]: Loading plugins:
2016-11-09 18:16:53 [INFO:root]:         C:\Users\all\Documents\Microdrop\plugins\dmf_control_board_plugin
2016-11-09 18:17:04 [INFO:root]: Traceback (most recent call last):
  File "C:\Users\all\MicroDrop\lib\site-packages\microdrop\plugin_manager.py", line 56, in load_plugins
    exec(import_statement)
  File "<string>", line 1, in <module>
  File "C:\Users\all\Documents\Microdrop\plugins\dmf_control_board_plugin\__init__.py", line 30, in <module>
    from feedback import (FeedbackOptions, FeedbackOptionsController,
  File "C:\Users\all\Documents\Microdrop\plugins\dmf_control_board_plugin\feedback.py", line 38, in <module>
    from matplotlib.backends.backend_gtkagg import (FigureCanvasGTKAgg as
  File "C:\Users\all\MicroDrop\lib\site-packages\matplotlib\backends\backend_gtkagg.py", line 14, in <module>
    from matplotlib.backends.backend_gtk import gtk, FigureManagerGTK, FigureCanvasGTK,\
  File "C:\Users\all\MicroDrop\lib\site-packages\matplotlib\backends\backend_gtk.py", line 36, in <module>
    from matplotlib.backends.backend_gdk import RendererGDK, FigureCanvasGDK
  File "C:\Users\all\MicroDrop\lib\site-packages\matplotlib\backends\backend_gdk.py", line 33, in <module>
    from matplotlib.backends._backend_gdk import pixbuf_get_pixels_array
ImportError: No module named _backend_gdk

2016-11-09 18:17:04 [ERROR:root]: Error loading dmf_control_board_plugin plugin.
Traceback (most recent call last):
  File "C:\Users\all\MicroDrop\lib\site-packages\microdrop\plugin_manager.py", line 56, in load_plugins
    exec(import_statement)
  File "<string>", line 1, in <module>
  File "C:\Users\all\Documents\Microdrop\plugins\dmf_control_board_plugin\__init__.py", line 30, in <module>
    from feedback import (FeedbackOptions, FeedbackOptionsController,
  File "C:\Users\all\Documents\Microdrop\plugins\dmf_control_board_plugin\feedback.py", line 38, in <module>
    from matplotlib.backends.backend_gtkagg import (FigureCanvasGTKAgg as
  File "C:\Users\all\MicroDrop\lib\site-packages\matplotlib\backends\backend_gtkagg.py", line 14, in <module>
    from matplotlib.backends.backend_gtk import gtk, FigureManagerGTK, FigureCanvasGTK,\
  File "C:\Users\all\MicroDrop\lib\site-packages\matplotlib\backends\backend_gtk.py", line 36, in <module>
    from matplotlib.backends.backend_gdk import RendererGDK, FigureCanvasGDK
  File "C:\Users\all\MicroDrop\lib\site-packages\matplotlib\backends\backend_gdk.py", line 33, in <module>
    from matplotlib.backends._backend_gdk import pixbuf_get_pixels_array
ImportError: No module named _backend_gdk
```

## Steps to reproduce the error ##

1. Install [MicroDrop v2.1](https://github.com/wheeler-microfluidics/microdrop/releases/tag/v2.1).
2. Launch the `MicroDrop environment prompt` link.
3. Run the command `ipython` to start an IPython shell.
4. Execute the command `from matplotlib.backends.backend_gtkagg import FigureCanvasGTKAgg`.

ryanfobel, 2016-11-10:
It looks like matplotlib is installed from the default channel; i.e., the output of `conda list` is:

```
matplotlib                1.5.3               np111py27_1
```

Replacing this version of `matplotlib` with the one from `wheeler-microfluidics` channel using the command:

```
conda install -c wheeler-microfluidics matplotlib
````

resolves the problem.

cfobel, 2016-11-10:
This is fixed as of b97cd07, and [release 2.2][1]:

The fix includes the following changes to MicroDrop:

 * 4bf2104 - **[FIX] Add wheeler channel to installed Conda**
 * 841f8b2 - **[FIX] Fix order of channels, wheeler 1st priority**

It also required rebuilding the [`wheeler-microfluidics/matplotlib` Conda package][2] using this attached Conda build recipe: [matplotlib-1.5.3.zip](https://github.com/wheeler-microfluidics/microdrop/files/584164/matplotlib-1.5.3.zip)

[1]: https://github.com/wheeler-microfluidics/microdrop/releases/tag/v2.2
[2]: https://anaconda.org/wheeler-microfluidics/matplotlib

closed 2016-11-10 by cfobel