sci-bots/microdrop #226: Building installer sometimes includes packages from the wrong channel

ryanfobel, 2017-05-26:
# Problem

When building a MicroDrop install using the command `paver build_installer`, [`constructor`](https://github.com/conda/constructor) downloads packages from the channels listed in [`.miniconda-recipe/construct.yaml`](https://github.com/wheeler-microfluidics/microdrop/blob/f6a42f5f82136b9f35b83b7f1cf4d49e97d848ba/.miniconda-recipe/construct.yaml#L4-L6). The order of channels in this list provides a mechanism for prioritizing packages from a certain channel (e.g., we need to use the `matplotlib` package from the [`wheeler-microfluidics`](https://anaconda.org/wheeler-microfluidics/matplotlib) channel versus the one from the [`conda-forge`](https://anaconda.org/conda-forge/matplotlib) channel because we require the GTK backend). A problem arises when a newer version of the package is released in the lower priority channel. For example, if the version of `matplotlib` in the `wheeler-microfluidics` and `conda-forge` channels are `2.0.0` and `2.0.2` respectively, the newer package (i.e., `2.0.2` from `conda-forge`) will take precedence even if it is the lower priority channel. Therefore, using the order of channels to prioritize certain packages is brittle to updates in other channels that are beyond our control.

# Possible workarounds

## 1. Pin versions in construct.yaml

If the version number for a given package is pinned in [`.miniconda-recipe/construct.yaml`](https://github.com/wheeler-microfluidics/microdrop/blob/dev/.miniconda-recipe/construct.yaml), it should be robust to updates in other channels.

## 2. Only update from compatible channels

This provides better control over which packages may end up in a MicroDrop install but places an additional burden on us to maintain packages for all of our dependencies.

## 3. Other options???

cfobel, 2017-05-26:
# Opting out of package updates during MicroDrop triggered package installs

Once a MicroDrop Miniconda environment is installed, **packages should only
ever be installed/updated when triggered to do so through the MicroDrop
GUI**.

This includes, e.g.:

- Any provided mechanism for auto-updating to the latest version of
MicroDrop
- Installing new plugins

Conda provides support for installing packages **without updating
dependencies** (see [here for details][1]).  If **all install operations
performed by MicroDrop utilize the `--no-update-dependencies` flag** no
packages will be updated unless explicitly required by the new package.

## Example

1. MicroDrop installer created using `constructor`, includes `matplotlib`
from `wheeler-microfluidics` with GTK support.
2. A newer version of `matplotlib` is published on the `conda-forge`
channel.
3. A new plugin is installed using the MicroDrop GUI (which uses the
`--no-update-dependencies` flag).
4. The `matplotlib` package **is not updated**.

[1]:
https://stackoverflow.com/questions/36195836/why-does-conda-try-to-update-packages-with-no-update-dependencies#

cfobel, 2017-05-26:
**Note: the only reason why MicroDrop cannot use the mainline `matplotlib`
package is that the GUI currently relies on GTK.  If/when MicroDrop drops
dependency on GTK, the `matplotlib` package should no longer be an issue.**