sci-bots/microdrop #228: Continuous integration strategy for Conda packages

cfobel, 2017-05-30:
# Conda packaging

## End goal

**Each Conda package** in any **Sci-Bots related Conda package channels (e.g., `wheeler-microfluidics`, `microdrop-plugins`)** should be set up with the following continuous integration workflow:

 1. Windows Python 2.7 32-bit (Appveyor):
  * Build Conda package
  * Run unit tests
 2. Linux Python 2.7 32-bit (Travis):
  * Build Conda package
  * Run unit tests

**Caveats:** We already plan to deprecate some packages, so they will not need to be handled.  Ryan and I will gladly go over a list of packages to indicate which packages can be skipped.

## Sub tasks

 1. [ ] **(1)** Implement Windows Python 2.7 32-bit continuous workflow for exemplary pure Python Conda package, `jupyter-helpers`
 2. [ ] **(2)** Implement Windows Python 2.7 32-bit continuous workflow for exemplary Python Conda package including C-extension, **TBD**
 3. [ ] **(3)** Write documentation on how to set up Appveyor CI workflow for a Conda package
 4. [ ] **(4)** Develop tool to automatically generate Appveyor configuration (according to documented steps from above) from a cloned working copy of a `git` repo
 5. [ ] **(5)** Export JSON-formatted info on all packages from Sci-Bots related Conda package channels (e.g., `wheeler-microfluidics`, `microdrop-plugins`)
         * Can use `conda search --override-channels -c <channel name> --json`
 6. [ ] **(6)** Develop tools accepting JSON format from `conda search` to:
     1. [ ] **(6.1)** Identify packages missing `["about"]["url"]` field (should point to corresponding GitHub page)
     2. [ ] **(6.2)** Identify packages where `["about"]["url"]` field corresponds to Sci-Bots managed `git` repo (i.e., belonging to either `wheeler-microfluidics` or `sci-bots` GitHub organizations)
     3. [ ] **(6.3)** Clone a `git` repo and execute tool **4** to generate corresponding Appveyor config
 7. [ ] **(7)** Apply **6.3** to each package identified using **6.2**.

Lucaszw, 2017-06-06:
# CI Workflow for AppVeyor + Conda

## Setting Up Conda Authentication Token

Before uploading to Anaconda Cloud, you must make an authentication token for your conda account

```python
# Login to your account:
> anaconda login
#Using Anaconda API: https://api.anaconda.org
#Username: lucaszw
#lucaszw's Password:
#login successful

# Generate an authentication token
> binstar auth -n someNameForToken --max-age 22896000 -c --scopes api
```

Enter the token above using the AppVeyor UI located here:

https://ci.appveyor.com/project/USERNAME/PROJECTNAME/settings/environment

Select Environment Variables > Add Variable:

```
| Name          | Value                |           
| ------------- |----------------------|
| binstar_token | AUTHENTICATION_TOKEN |
```

## Sample YAML configuration for jupyter_helpers

```yaml
# Set environment to include Miniconda, Python 2.7, and 32 bit architecture
# for other python versions and architectures simple add another entry in the matrix
environment:
  matrix:
    - PYTHON_VERSION: 2.7
      MINICONDA: C:\Miniconda
      PYTHON_ARCH: "32"

# Specify a version number based on the build
version: '1.0.{build}'

# On initiation print out which python version and build number to simplify debugging
init:
  - "ECHO %PYTHON_VERSION% %MINICONDA%"

install:
  # Add conda to the path
  - "set PATH=%MINICONDA%;%MINICONDA%\\Scripts;%PATH%"
  # Activate the conda environment
  - "call %MINICONDA%\\Scripts\\activate.bat"
  # Configure conda to ensure that it automatically answers yes to promps
  - conda config --set always_yes yes --set changeps1 no
  # Add conda-forge and wheeler-microfluidics channels
  - conda config --add channels conda-forge
  - conda config --add channels wheeler-microfluidics
  # Update conda, and install build tools, and nose, and other dependencies
  - conda update -q conda
  - conda install --yes conda-build jupyter path_helpers nose
  # Print out conda indo to simplify debugging
  - conda info -a
  # Install pip dependencies (should migrate to conda later)
  - pip install psutil
  # Fetch the file location for conda-build
  - FOR /F "tokens=*" %%a in ('conda-build . --output') do SET PACKAGE_LOCATION=%%a
  - echo %PACKAGE_LOCATION%
  # Build package
  - conda-build .
  # Upload built package
  - binstar -t %BINSTAR_TOKEN% upload --force %PACKAGE_LOCATION%

build: false

test_script:
  # Test using nosetests, and export the test results to and xml file (to upload to AppVeyor)
  - nosetests .\\jupyter_helpers\\tests -vv --with-xunit

after_test:
  # After tests, use powershell to upload the test results (note, xunit for AppVeyor appears to   #be different than xunit for nose)
  - ps: $wc = New-Object 'System.Net.WebClient'
  - ps: $wc.UploadFile("https://ci.appveyor.com/api/testresults/junit/$($env:APPVEYOR_JOB_ID)", (Resolve-Path .\nosetests.xml))
```

## Common config problems

1. Ensure the build phase is turned off for python development

2. Ensure the environment is activated

3. For debugging connect using Remote Desktop

4. Subprocess' need to be physically terminated at end of test scripts (otherwise the AppVeyor process will hang). To do this:

   1. Ensure that Subprocesses are not flagged to be terminated with main subprocess

   2. Add a stop method to the Thread, and call it at the end of the tests

   3. Use PSutil to terminate process, since this will ensure that Windows doesn't open a confirmation prompt (which causes that AppVeyor tests to hang 

      ```python
      import psutil
      ```

      ```python
      def test_get_session():
          sm = notebook.SessionManager()
          session = sm.get_session()
          session.thread.stop()
          process = psutil.Process(session.process.pid)
          for proc in process.children(recursive=True):
              proc.kill()
          process.kill()
          return session
      ```

## Sample Code

```python
class StoppableThread(Thread):
    """Thread class with a stop() method. The thread itself has to check
    regularly for the stopped() condition.
    From: https://stackoverflow.com/questions/323972/is-there-any-way-to-kill-a-thread-in-python"""

    def __init__(self, *args, **kwargs):
        super(StoppableThread, self).__init__(*args,**kwargs)
        self._stop_event = threading.Event()

    def stop(self):
        self._stop_event.set()

    def stopped(self):
        return self._stop_event.is_set()
```

```python
def enqueue_output(out, queue):
    """ Ensure break statement in StoppableThread target """
    for line in iter(out.readline, b''):
        if self.thread.stopped(): break
            queue.put(line)
            out.close()
```

## Resources

1. Tutorial for handling Conda with AppVeyor: http://tjelvarolsson.com/blog/how-to-continuously-test-your-python-code-on-windows-using-appveyor/
2. Ensuring termination of subprocesses before main process: https://stackoverflow.com/questions/4789837/how-to-terminate-a-python-subprocess-launched-with-shell-true/25134985
3. Uploading test results to AppVeyor: https://www.appveyor.com/docs/running-tests/#uploading-xml-test-results
4. Ensuring thread is not killed abruptly: https://stackoverflow.com/questions/323972/is-there-any-way-to-kill-a-thread-in-python
5. Support for figuring out why nosetests was hanging on AppVeyor: http://help.appveyor.com/discussions/problems/6777-appveyor-with-pythonminiconda-hanging-after-completing-nosetests
6. How to connect to Remote Desktop for AppVeyor instance: https://www.appveyor.com/docs/how-to/rdp-to-build-worker/

Lucaszw, 2017-06-06:
Will make a PR once all tasks are complete, but here is the fork I used for the above.
[Jupyter Helpers with CI Workflow](https://github.com/Lucaszw/jupyter-helpers)

Lucaszw, 2017-07-14:
[Updated with Sci-Bots-Configs Workflow](https://github.com/sci-bots/sci-bots-configs)