sci-bots/microdrop #230: Importing electrodes with irregular geometries

ryanfobel, 2017-06-12:
MicroDrop sometimes fails when importing device designs that have electrodes with non-standard geometries. For example, [this device](https://github.com/wheeler-microfluidics/microdrop/files/1069061/NMR-DMF.Coil.v1.01.svg.zip) has a spiral-shaped electrode and fails with the following traceback:

```python
Traceback (most recent call last):
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\microdrop\gui\dmf_device_controller.py", line 222, in load_device
    **kwargs)
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\microdrop\dmf_device.py", line 61, in load
    return cls(svg_filepath, **kwargs)
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\microdrop\dmf_device.py", line 77, in __init__
    svg_canvas = ShapesCanvas(self.df_shapes, self.shape_i_columns)
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\svg_model\shapes_canvas.py", line 54, in __init__
    shape_i_columns)
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\svg_model\tesselate.py", line 24, in tesselate_shapes_frame
    triangulator = Triangulator(df_path[['x', 'y']].values)
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\svg_model\seidel.py", line 213, in __init__
    self.process()
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\svg_model\seidel.py", line 230, in process
    traps = self.query_graph.follow_edge(edge)
  File "C:\Users\Ryan\MicroDrop\lib\site-packages\svg_model\seidel.py", line 511, in follow_edge
    while(edge.q.x > trapezoids[-1].right_point.x):
AttributeError: 'NoneType' object has no attribute 'right_point'
```

cfobel, 2017-06-13:
Perhaps you could try the workaround described in #222 ?

cfobel, 2017-06-20:
# Cause

The problem is due to [issue 2 in the `svg-model` package][1].

# Workaround

While the updated `svg-model` package will be included in the next MicroDrop installer release, to fix an existing MicroDrop installation, perform the following steps:

 1. Launch MicroDrop Environment Prompt from Windows Start menu.
 2. Execute the following command (note the sci-bots-unstable Conda channel):

        conda install -c wheeler-microfluidics -c conda-forge "svg-model>=0.9" "dmf-device-ui>=0.3"

# Device including connections

I've attached a copy of the device from the original post above including connections between adjacent electrodes:
 
[NMR-DMF Coil v1.01.zip](https://github.com/wheeler-microfluidics/microdrop/files/1089082/NMR-DMF.Coil.v1.01.zip)

# TODO

**Bump required versions of `svg-model` and `dmf-device-ui` to `"svg-model>=0.9" "dmf-device-ui>=0.3"`**. 

[1]: https://github.com/wheeler-microfluidics/svg-model/issues/2