sci-bots/microdrop #264: bug(microdrop): actuation errors reference electrode IDs not channel numbers

ryanfobel, 2018-11-01:
Currently, when there is an actuation error, the error dialog reports electrode IDs. This is not very useful to the user as it is difficult to map electrode IDs to the device view (electrode IDs are only shown to the user when hovering over a specific electrode).

[image]

I suggest that these error dialogs should show channel IDs, since these can be displayed on the device view using the "channel_labels" layer.

cfobel, 2018-11-01:
In my mind, **_electrodes_** are what are being actuated (specific to the DMF chip, but _not_ specific to the DropBot; e.g., the DropBot plugin may not even be installed) and **_channels_** are a DropBot implementation detail.  @ryanfobel would you agree with this?[<sup>1</sup>](#1)

That said, since we are only really concerned with the DropBot, I certainly agree that the  _electrode IDs_ are not particularly useful.  However, this could be addressed by simply changing the `id` attribute of each electrode path in the SVG (i.e., the _electrode IDs_) to match the respective DropBot channel number.  @ryanfobel what do you think?

-----------------------------------

<a class="anchor" id="1">1.</a> In MicroDrop, all **actuation-related** code _outside of the DropBot plugin_ operates on _electrode IDs_.  The DropBot plugin then maps requests for actuation of electrodes to the corresponding hardware channels.

cfobel, 2019-01-25:
@ryanfobel and@cfobel decided to postpone matching of electrode IDs to DropBot channels until we finalize the new 120-channel chip layout.