sci-bots/microdrop #266: bug(microdrop): Device loading fails for devices with no connections

joshdahmer, 2019-01-11:
When attempting to load device file without any connections (such as a pin map) device load fails with the following traceback:

```
2019-01-11 15:56:00.529 [ERROR:microdrop.gui.dmf_device_controller.DmfDeviceController.load_device]: Error loading device.
Traceback (most recent call last):

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\microdrop\gui\dmf_device_controller.py", line 236, in load_device
    **kwargs)

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\microdrop\dmf_device.py", line 43, in load
    return cls(svg_filepath, **kwargs)

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\microdrop\dmf_device.py", line 104, in __init__
    self.electrode_neighbours = electrode_neighbours(self)

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\microdrop\dmf_device.py", line 496, in electrode_neighbours
    0].groupby('source')['y_delta'].idxmin()

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\pandas\core\generic.py", line 5162, in groupby
    **kwargs)

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\pandas\core\groupby.py", line 1848, in groupby
    return klass(obj, by, **kwds)

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\pandas\core\groupby.py", line 516, in __init__
    mutated=self.mutated)

  File "C:\Users\Wheeler\Miniconda2\envs\zika-dev\lib\site-packages\pandas\core\groupby.py", line 2934, in _get_grouper
    raise KeyError(gpr)
KeyError: 'source'
```

cfobel, 2019-01-14:
This should be fixed as of **version [2.32.2](https://github.com/sci-bots/microdrop/releases/tag/v2.32.2)**; specifically, the fix was introduced in dc849ad5e728524dbfd402a933280216c7111b2a.

Note that as of this post, **version [2.32.2](https://github.com/sci-bots/microdrop/releases/tag/v2.32.2)** is **NOT** available as a publicly published Conda package (version 2.19 is the most recent published version).

cfobel, 2019-01-14:
I've published a copy of `microdrop@2.32.2` to the `sci-bots/label/microdrop-2` channel.

It can be installed in an existing Conda environment using the following command:

```sh
conda install -c sci-bots/label/microdrop-2 -c sci-bots -c conda-forge microdrop
```

closed 2019-01-14 by cfobel